F2Wang/ObjectDatasetTools #3: make my own datasets for

hlcool, 2019-06-13:
Have you ever paid attention to [DenseFusion](https://github.com/j96w/DenseFusion) and [Deep_Object_Pose](https://github.com/NVlabs/Deep_Object_Pose)？

This two projects also realize 6D pose estimation. If I want to use the above two projects to train my data, can I use your tools to create my own datasets?

F2Wang, 2019-06-13:
Hi,

I haven't looked into those 2 projects, I wrote the codes originally to help myself prepare data for singleshotpose.

I do have plan to generate customized data in complaint with the YCB video dataset, so that it can be used with poseCNN and DenseFusion, I will let you know when I have done that.

Thanks,
Fan

hlcool, 2019-06-14:
Hey F2Wang：
    great job，very happy to hear what you said. Looking forward to your next job.
because I have a realsense D435 in my hand,  I will continue to use ObjectDatasetTools to prepare data for singleshotpose.
by the way，Do you usually use Wechat?

ghoshaw, 2019-06-17:
> Hey F2Wang：
> great job，very happy to hear what you said. Looking forward to your next job.
> because I have a realsense D435 in my hand, I will continue to use ObjectDatasetTools to prepare data for singleshotpose.
> by the way，Do you usually use Wechat?

Hi， Is the data generated by this tool good enough to train singleshotpose?

F2Wang, 2019-06-17:
> Hi， Is the data generated by this tool good enough to train singleshotpose?

It should have anything you need for singleshotpose.

F2Wang, 2019-06-17:
> Hey F2Wang：
> great job，very happy to hear what you said. Looking forward to your next job.
> because I have a realsense D435 in my hand, I will continue to use ObjectDatasetTools to prepare data for singleshotpose.
> by the way，Do you usually use Wechat?

Thank you, I am not on Wechat, but feel free to open any issue here if you have any problem using the code.

ghoshaw, 2019-06-18:
Hi, I tried to go through the whole pipeline. But when I  create label files , it says "trimesh object has no attribute 'apply_obb'", I reinstall the trimesh, but the error is still on, Do you have any idea? Thank you

hlcool, 2019-06-18:
> Thank you, I am not on Wechat, but feel free to open any issue here if you have any problem using the code.

OK, I'm trying now. 
Thanks for your effort

hlcool, 2019-06-18:
> Hi, I tried to go through the whole pipeline. But when I create label files , it says "trimesh object has no attribute 'apply_obb'", I reinstall the trimesh, but the error is still on, Do you have any idea? Thank you

If I use the timer images given by the author, I also encounter the same  @F2Wang @ghoshaw

F2Wang, 2019-06-18:
> > Hi, I tried to go through the whole pipeline. But when I create label files , it says "trimesh object has no attribute 'apply_obb'", I reinstall the trimesh, but the error is still on, Do you have any idea? Thank you
> 
> If I use the timer images given by the author, I also encounter the same problem。@F2Wang @ghoshaw

Hi, my apologies, this seems to be a trimesh version issue, install a newer trimesh version should solve the problem:
`sudo python -m pip install trimesh==2.38.24`
I will update the installation guide as well.

ghoshaw, 2019-06-19:
That works for me, Thank you very much!
And I have another question to bother you, In my understanding, after the step 5 in your guide, it runs trimesh.apply_obb to get the 3D bbox, but I do not know how the bbox is set.
https://raw.githubusercontent.com/F2Wang/ObjectDatasetTools/master/doc/setup.png
like this pic, if the sugar box is cylinder, then how can I make sure that between two records, the 3D bbox is the same, e.g. the sugar side face to the same 3D bbox side? Hope I made myself clear.

F2Wang, 2019-06-19:
> That works for me, Thank you very much!
> And I have another question to bother you, In my understanding, after the step 5 in your guide, it runs trimesh.apply_obb to get the 3D bbox, but I do not know how the bbox is set.
> https://raw.githubusercontent.com/F2Wang/ObjectDatasetTools/master/doc/setup.png
> like this pic, if the sugar box is cylinder, then how can I make sure that between two records, the 3D bbox is the same, e.g. the sugar side face to the same 3D bbox side? Hope I made myself clear.
> 
> [image]
> Even is one record, if I want the 3D bbox being set as shown above, not the others way. How can I make sure I get the desired 3D bbox?

Glad the apply_obb problem is solved!

I am not entirely sure If I understand you correctly so pardon me if I am not answering your question right. 

Currently, the apply_obb function applies a transform to the processed mesh from registered_scene.ply so that the transformed mesh has its AABB centered at the origin and the same dimensions as the OBB, this new transformed mesh will be saved as objectname.ply, open it in meshlab and save it as non-binary file, and then this objectname.ply should be used for Singleshotpose.

The first corner of the 3D bounding box is the [min_x, min_y, min_z] of the transformed mesh after apply_obb, the second corner is [min_x, min_y, max_z] ... etc. 2D projections of the eight 3D corners are computed on all images in the sequence. That being said,  this code does not support the 3D bounding box of a specified orientation, and it will be quite difficult to do that.

A video example may be helpful for explaining here, I uploaded a video of pose estimation with singleshotpose, the blue box is the result and the green box is the ground truth. 

https://www.youtube.com/watch?v=aqVZb1sMUxo

ghoshaw, 2019-06-19:
Thank you very much, that helps a lot!

CaffeMrDe, 2019-07-22:
Hi
I just download the code. And I find the question.
In Singleshotpose, dataset  requrie the 3D BBOX. But I saw the get_BBs.py. It only include the 2D points. So, how should I create  the  3D bounding box  in this project?

` 
x,y,w,h = cv2.boundingRect(cnt)
            cv2.rectangle(img_original,(x,y),(x+w,y+h),(0,255,0),2)
            data.append(str(x))
            data.append(str(y))
            data.append(str(x+w))
            data.append(str(y+h))
            data.append(str(0))
            allinfo.append(data)
`

Can you help me? Thanks

ghoshaw, 2019-07-23:
In create_label_files.py, you can find “points = mesh.bounding_box.vertices center = mesh.centroid” in line 86 and 87, that is the 3d points.

| |
ghoshaw
|
|
ghoshaw@163.com
|
签名由网易邮箱大师定制

CaffeMrDe, 2019-07-23:
Oh. Thanks.! I have already understand.

zhangxiaodi, 2019-07-24:
i use timer dataSet and found the generator the file of .ply have no faces attribute. And this lead the mesh still read error! So who can give me some advice? thanks!

F2Wang, 2019-07-25:
> i use timer dataSet and found the generator the file of .ply have no faces attribute. And this lead the mesh still read error! So who can give me some advice? thanks!

The generated .ply file is essentially registered pointclouds(vertices and no face), you will need to use 3rd party software to triangulate the mesh (e.g., meshlab, blender). You can refer to the readme section for more instruction on that.

MrLuer, 2019-09-11:
> 您是否注意过[DenseFusion](https://github.com/j96w/DenseFusion)和[Deep_Object_Pose](https://github.com/NVlabs/Deep_Object_Pose)？
> 
> 这两个项目也实现了6D姿态估计。如果我想使用上述两个项目来训练我的数据，我可以使用你的工具来创建我自己的数据集吗？

你好，我最近也在做6d姿态估计，请问用这里的工具能制作数据集用在densefusion上可以使用吗，谢谢

F2Wang, 2019-09-11:
Yes and no, It actually produces everything required by densefusion, but in a different format, and you will need to convert your data to the format dense fusion requires, which is not trivial.

That being said, I have always been wanting to support dense fusion, but don't have time to do so in the near future, I will really appreciate if someone can help contribute.

MrLuer, 2019-09-24:
@F2Wang 
to support  the densefusion，need to generate rotation and translation matrix，but I have no ideal to translate the 21 dimensional data to rotation and translation matrix.

ghoshaw, 2019-09-24:
I guess solvepnp is what you need.

MrLuer, 2019-09-24:
thank you very much，it's helpful a lot

YanqingWu, 2019-10-08:
@MrLuer 老哥，能加个微信吗？我在densefusion也看到你了。

YanqingWu, 2019-10-08:
@hlcool 老哥，你也在做densefusion吗，能加个微信交流一下吗。

MrLuer, 2019-10-08:
可以啊，我微信 18916589327

jhellzhang, 2019-10-18:
> Have you ever paid attention to [DenseFusion](https://github.com/j96w/DenseFusion) and [Deep_Object_Pose](https://github.com/NVlabs/Deep_Object_Pose)？
>
> This two projects also realize 6D pose [...]

have you another link about the "preprocessed LINEMOD dataset" for "singleshotpose"?because i can use google

kunkunli, 2019-12-18:
> Glad the apply_obb problem is solved!
>
> I am not entirely sure If I understand you correctly so pardon me if I am not answering your question right.
>
> Currently, the apply_obb function applies a [...]

 How do you solve the apply_obb problem？Thanks.

closed 2021-05-18 by hlcool