F2Wang/ObjectDatasetTools #4: error occured while read png with python3

Sciroccogti, 2019-07-22:
It seems to be incompatible with *python3*.

I choose to read `.png` with *PIL* and seemingly repaired it.

```Python
reader = png.Reader(depth_file)
pngdata = reader.read()
depth = np.array(map(np.uint16, pngdata[2]))
```
swap all the fragments above with this:
```Python
depth = np.array(Image.open(depth_file))
```

Meanwhile, you need to typecast npcds to int:
```Python
int(npcds)
```

zhangxiaodi, 2019-07-24:
hi, Maybe  depth = np.vstack(map(np.uint16, pngdata[2])) to replace depth = np.array(Image.open(depth_file)).     

npcds = int(npcds)
is there have any meeting in offline?

Sciroccogti, 2019-07-24:
@zhangxiaodi Thanks for your suggestion!

However, I have already issued a [PR](#6 ) which seems to have fixed Python incompatibleness. I noticed that your method don't require extra module which is pretty cool.

Nope, there's no offline meetings(how could it be!), sadly😁.

F2Wang, 2019-07-25:
> @zhangxiaodi Thanks for your suggestion!
> 
> However, I have already issued a [PR](#6) which seems to have fixed Python incompatibleness. I noticed that your method don't require extra module which is pretty cool.
> 
> Nope, there's no offline meetings(how could it be!), sadlygrin.

Thank you for the suggestion, I am currently on vacation and don't have a linux computer with me, I will look more closely on this once I am back.

commit dd478d9f (2019-07-28): fixed #4
closed 2019-07-28 by F2Wang