F2Wang/ObjectDatasetTools #5: Error occurs in step 6

Sciroccogti, 2019-07-23:
Thanks a lot for your Tools!

However, have I completed steps 1 to 5, an error occurred when I try to create label files. Here is the error:
```Bash
$ python3 create_label_files.py LINEMOD/lays
lays is assigned class label 0.
Traceback (most recent call last):
  File "create_label_files.py", line 96, in <module>
    mesh = trimesh.load(folder + "registeredScene.ply")
  File "/usr/local/lib/python3.5/dist-packages/trimesh/exchange/load.py", line 135, in load
    **kwargs)
  File "/usr/local/lib/python3.5/dist-packages/trimesh/constants.py", line 137, in timed
    result = method(*args, **kwargs)
  File "/usr/local/lib/python3.5/dist-packages/trimesh/exchange/load.py", line 205, in load_mesh
    **kwargs)
  File "/usr/local/lib/python3.5/dist-packages/trimesh/exchange/ply.py", line 76, in load_ply
    ply_ascii(elements, file_obj)
  File "/usr/local/lib/python3.5/dist-packages/trimesh/exchange/ply.py", line 485, in ply_ascii
    row = array[position]
IndexError: index 76862 is out of bounds for axis 0 with size 76862
```

I have also tried with python2.7, but the error appeared to be the same.

If the `.ply` is saved in **ASCII**, error would change into: 
```Bash
lays is assigned class label 0.
Traceback (most recent call last):
  File "create_label_files.py", line 96, in <module>
    mesh = trimesh.load(folder + "registeredScene.ply")
  File "/usr/local/lib/python3.5/dist-packages/trimesh/exchange/load.py", line 135, in load
    **kwargs)
  File "/usr/local/lib/python3.5/dist-packages/trimesh/constants.py", line 137, in timed
    result = method(*args, **kwargs)
  File "/usr/local/lib/python3.5/dist-packages/trimesh/exchange/load.py", line 205, in load_mesh
    **kwargs)
  File "/usr/local/lib/python3.5/dist-packages/trimesh/exchange/ply.py", line 78, in load_ply
    ply_binary(elements, file_obj)
  File "/usr/local/lib/python3.5/dist-packages/trimesh/exchange/ply.py", line 586, in ply_binary
    populate_listsize(file_obj, elements)
  File "/usr/local/lib/python3.5/dist-packages/trimesh/exchange/ply.py", line 551, in populate_listsize
    dtype=field_dtype)[0]
IndexError: index 0 is out of bounds for axis 0 with size 0
```

I really need some help, cause I have a deadline on July 31st💀

Sciroccogti, 2019-07-23:
Sorry, I skipped your tutorial on Youtube which turns out to be quite helpful. Now everything is OK!

closed 2019-07-23 by Sciroccogti

kevinchan04, 2020-01-08:
how to solve this , share the solution with details please ,thanks a lot!