F2Wang/ObjectDatasetTools #7: Can I use it for multi obj?

Sciroccogti, 2019-07-28:
I tried to directly record pictures with four items in it, but the `.ply` ,especially the occluded sides of the items, is quite bad.

F2Wang, 2019-07-28:
Sorry, the code current cannot handle multi objects in a scene.

Is there any particular reason you want to record multiple objects at the same time? If you want to train object detector/pose estimator to be robust against other objects in the background you can add a data augmentation step for your training data.

Also could you elaborate on how the registered pointcloud is bad for multi objects in the same scene? There will be occluded part for sure, since part of the object will be invisible to the camera throughout the entire scan, but is the alignment also bad? 

The reason why I asked is that I have been considering supporting multiple objects in the same scene, but I am uncertain if such functionalities will be useful in the first place.

closed 2019-12-07 by Sciroccogti

Sciroccogti, 2019-12-07:
Sorry for the late reply.

I was in a competition which requires us to recognize multiple objects on a table, and that's why I wanted to use the tool to record multiple objects and make a dataset for yolo6d.

And I'm sorry that I haven't dived into the algorithm, so I can hardly understand the technical words you've said... I'll be back as long as I've got further knowledge of these things.

Anyway, thanks a lot for your repo! It helped us to win a national prize!

elevenjiang1, 2020-10-17:
So,are there any tools that can make multi objects?
I want to make a dataset like YCB Video Dataset(https://rse-lab.cs.washington.edu/projects/posecnn/),but it is very difficult to do so.

kiterunnerly, 2021-03-02:
Excuse me, did you solve the problem?how to use the tool to record multiple objects?

Sciroccogti, 2021-03-02:
> Excuse me, did you solve the problem?how to use the tool to record multiple objects?

No, I didn't solved it. We just made single object datasets, as the yolo6d seemed to be powerful enough to go without multi object datasets.