F2Wang/ObjectDatasetTools #8: AttributeError: 'NoneType' object has no attribute 'shape' while creating Image Masks

AkmaralMoldagalieva, 2019-08-04:
Hello! I am Akmaral. 
First of all, thank you for this great job and sharing with us. 

I am getting this error:  " AttributeError: 'NoneType' object has no attribute 'shape'
"  while running  'python3 create_label_files.py LINEMOD/xxx' (Step 6).  I have successfulIy  completed all previous steps, and have registeredScene.ply inside my folder. Even updating my trimesh version doesn't help. Please can you give me a hint to solve this problem.

AkmaralMoldagalieva, 2019-08-04:
I could solve this problem. I just changed the .ply file, used not registeredScene.ply but the .ply created manually in step 5. 

Thank you and sorry for troubling! ) Hopefully, I will get good results.

closed 2019-08-04 by AkmaralMoldagalieva

F2Wang, 2019-08-04:
Good luck! And feel free to open another issue if you encounter any other problem.

sanjaysswami, 2019-09-27:
@AkmaralMoldagalieva did you solve it?

AkmaralMoldagalieva, 2019-09-27:
Hello!
The problem might be that you didn't change 'registeredScene.ply' file in create_labels.py file (line 96). Please change this file name with the one that you've generated in step 5. 
Initially this generated .ply file has 'registeredScene.ply' name. After processing save it with another name and put this name into 96 line in  create_labels.py file. These steps solved in my case.

Feel free to write in case I need a help

sanjaysswami, 2019-09-27:
Hey thank you for the advice. I just followed as you said in your previous comments. I solved it.

8umpk1n, 2022-11-08:
我这么做了，可是仍然无法解决