F2Wang/ObjectDatasetTools #10: About the marker

ghoshaw, 2019-08-19:
Hi, you said "Color print the pdf with the correctly sized aruco markers in the arucomarkers folder.",  is that mean print the pdf in a A4 paper? because the object I reconstructed is not right in scale.

F2Wang, 2019-08-19:
Hi,

The marker size has no effect with the scaling of the reconstructed mesh, also all transforms computed with the code should be rigid  transforms with so scaling so there should not be a scaling issue at all. 

Could you provide more details on how the mesh is not right in scale? If it’s too small than you have excepted it might because it’s unit is in meter.

ghoshaw, 2019-08-19:
Thanks for your reply, I will check about the scale again.  Another question is, in camera.py , "depth_image = depth_image*1.0/65535*8", what the parameter means?

F2Wang, 2019-08-19:
Hi,

For realsense cameras the depth reading is coded in uint 16 interpolated over a 8m range, hence the line of code when I transform a depth reading to an actual depth in meter. That does remind me that if you are not using an realsense camera, it could be causing a scale issue. 
If you are okay with sending me your data for me to look at it, please email it to me at faninedinburgh@gmail.com.

Thanks,
Fan

ghoshaw, 2019-08-27:
Hi， F2Wang
     I tested the new code, and the scale problem is solved, Thank you!

danieldimit, 2019-09-04:
@F2Wang Sorry to bother you. You've written "Color print the pdf". As I am printing the markers a lot bigger it would be more cost efficient to print them black and white. Would there be a problem that the markers are not blue, like in the pdf that you've provided? Shouldn't it even be better if the markers are black and white, because of the additional contrast?

F2Wang, 2019-09-05:
> @F2Wang Thank you for the cool tool that you've made. You've written "Color print the pdf". As I am printing the markers a lot bigger it would be more cost efficient to print them black and white. Would there be a problem that the markers are not blue, like in the pdf that you've provided? Shouldn't it even be better if the markers are black and white, because of the additional contrast?

Hi, the reason for printing the marker blue is because realsense cameras don't see black item very well, and the code needs depth readings of the marker.