F2Wang/ObjectDatasetTools #14: About the register_segmented.py

ghoshaw, 2019-09-05:
Hi, I tried to use the script register_segmented.py  and adjust the MAX_RADIUS as you asked. But error happend:
Load and segment frames
  0%|          | 0/76 [00:00<?, ?it/s]/usr/local/lib/python2.7/dist-packages/numpy/core/fromnumeric.py:3118: RuntimeWarning: Mean of empty slice.
  out=out, **kwargs)
/usr/local/lib/python2.7/dist-packages/numpy/core/_methods.py:85: RuntimeWarning: invalid value encountered in double_scalars
  ret = ret.dtype.type(ret / rcount)

Traceback (most recent call last):
  File "register_segmented.py", line 216, in <module>
    originals = load_pcds(path, downsample = False, interval = RECONSTRUCTION_INTERVAL)     
  File "register_segmented.py", line 126, in load_pcds
    distance = point_to_plane(depth,sol)
  File "/home/ghoshaw/usr/Detect3D/ObjectDatasetTools/utils/plane.py", line 55, in point_to_plane
    plane_xyz = p[0:3]
TypeError: 'NoneType' object has no attribute '__getitem__'

If I run  register_scene.py, it was ok.
Is there any other parameter need to change or something? Thanks!

F2Wang, 2019-09-05:
Hi,

register_Segmented is an empirical script that uses some "hacky" method to remove the plane and noise greater than defined in the MAX_RADIUS. It's very likely that the script will not work, or yield some poor result.

The problem you described seems to be relating to the plane equation cannot found in some frames of your dataset due to not having enough QR markers within the scene (e.g. less than 2 complete markers).  A quick hacky way to try to get around this is to adjust your RECONSTRUCTION_INTERVAL, so that it will pick up a different set of frames to register that hopefuly all have enough markers in them, but I think it will be easier for you to just use register_Scene.

ghoshaw, 2019-09-06:
ok, thanks!

closed 2019-09-06 by ghoshaw