F2Wang/ObjectDatasetTools #17: To check ground truth and mesh generated

sanjaysswami, 2019-09-26:
@F2Wang I was trying to project back the ground truth generated but somehow it is not correct. Can you please tell me how I can project the ground truth back onto the RGB image and check it is correctly generated or not?
Or if you have any other idea to cross check can you please tell me?

I am trying to generate real data set and use it to train Dense fusion.

F2Wang, 2019-09-26:
Hi this should be automatically generated after running createlabelfiles, you can inspect it by running inspectMasks.py

sanjaysswami, 2019-09-27:
@F2Wang I have downloaded the sample dataset of timer that you have provided but it does not have depth scale. I have assumed it to 0.001 and I have generated the gt file for it. When i was trying to generate registered mesh file i got following error.

[image]

YanqingWu, 2019-10-08:
@sanjaysswami can this repo create the Rotation and transnational matrix?

sanjaysswami, 2019-10-08:
> @sanjaysswami can this repo create the Rotation and transnational matrix?

YES. The ground truth pose is always in terms of rotational and translation matrices

ghoshaw, 2019-11-15:
@sanjaysswami  But the value of the rotational and translation matrices in this repo is not equal to the PNP results. I tried to solvePnP using the the 2D point in image and 3D point, but the result is not the same as the npy saved in this repo.

zhangxiaodi, 2020-01-09:
> @sanjaysswami But the value of the rotational and translation matrices in this repo is not equal to the PNP results. I tried to solvePnP using the the 2D point in image and 3D point, but the result is not the same as the npy saved in this repo.

Maybe the result should be converted by np.T, I transformed those results into 8 corners which is the objects' control point and find the result is not correct. But I finally get the correct result after transformed the result matrix.

nihar0602, 2021-04-10:
Hi @sanjaysswami 
were you able to run DenseFusion using the dataset created via this? 
Also, did you made a new object dataset for this? 

Thanks!

YanqingWu, 2021-04-16:
> Hi @sanjaysswami
> were you able to run DenseFusion using the dataset created via this?
> Also, did you made a new object dataset for this?
> 
> Thanks!

I have tried, the answer is yes. But success rate is very low. And DenseFusion works poorly in the generated dataset.

nihar0602, 2021-04-22:
@YanqingWu 

Thank you very much for responding back to my query. 
Have you tried running it on SingleShotPose that as suggested by the author? 

Thanks again!

YanqingWu, 2021-04-23:
> @YanqingWu
> 
> Thank you very much for responding back to my query.
> Have you tried running it on SingleShotPose that as suggested by the author?
> 
> Thanks again!

Sorry, i did not try SinglePose.