F2Wang/ObjectDatasetTools #18: Parameter or aruco makers adjustment on big Objects

wangyidong3, 2019-10-07:
Hi
I want to get big object datasets' masks and labels, such as luggage, but after registering scene computation, the point clouds look like this:
[image]

I also tried to scan longer time, or use 9*4   aruco makers but get worse result.
Is there any parameter or the size of  aruco makers need to adjust?

If I manually rotate the object in Meshlab, it seems the masks are not match with the jpg files. Is it a correct way to do that?

Thanks.

F2Wang, 2019-10-08:
The setup appears to be right but the registration seems really off, I haven't tried registering any large item so far, do you mind sending me the original dataset to faninedinburgh@gmail.com? Thanks!

wangyidong3, 2019-10-20:
Hi Fan
Thank you so much for your suggestions.  
"(This tool) was designed only to reconstruct scene where the object of interest and other unwanted background moves in coherent motion (e.g., the camera rotates while the object of interest remains stable)"
I used turning table approach to collect the data, it made the algorithm prioritized matching the background instead of the object of interest. 
After I rotated camera instead of rotated luggage, the registration result is much better.
[image]
Thank you for your help.

closed 2019-10-20 by wangyidong3