F2Wang/ObjectDatasetTools #20: I can't generate the correct mesh for 'timer'

GraceJary, 2019-10-08:
I follow the instruction to construct the model for timer.
But I get a ply file like this.
[image]

F2Wang, 2019-10-08:
This is very strange ... did you change any parameters of the code?

GraceJary, 2019-10-08:
Yes, I change the instrinsics.json file under timer folder.
I add "depth_scale=1.0" to the instrinsics file.

F2Wang, 2019-10-10:
HI, the depth scale is wrong, it should be 0.00012498664727900177 for that particularly dataset, sorry about that.

GraceJary, 2019-10-11:
Ｔｈank you very much for the help. Now, I can get a mesh of the timer. But the result still looks with some error, the timer seems to be divided to two part.
[image]

F2Wang, 2019-10-11:
Hi, If you run register_Segmented this is mostly the expected behavior because the code projects the visible pointcloud to the plane (the missing side) to make it easier for reconstruction later. If you run register_Scene the code will not do this.

The timer is not a great example as the sides are not complete, the purpose of the sequence is to give an example of how to prepare a customized data without using realsense cameras.

jarygrace, 2019-10-12:
Would the order of these markers affect the result of compute pose? I don't know whether my markers are placed correctly.

F2Wang, 2019-10-12:
The order and the encodings of the aruco markers shouldn't affect the registration result at all.

F2Wang, 2019-10-15:
Please feel free to open another issue if you encounter other problems

closed 2019-10-15 by F2Wang