F2Wang/ObjectDatasetTools #23: to get highter precision dataset

MrLuer, 2019-10-23:
Thank you for your excellent work.
I got unsatisfactory preformance when training signpose with my own dataset. I guess it may be related to data annotation, I want you to give me some advice
1. is there any skills for data annotation? Such as cropping a mesh file
2.  except record.py and record2.py, have you tried to create datasets with higher precision parameters using realsense, such as with higher resolution?

F2Wang, 2020-01-04:
I hope your problem has been resolved by now. I am not sure what you mean by having unsatisfactory performance when training singleshotpose with your own dataset, do you mean that you have a low train/validation accuracy? That shouldn't be the case if you initialized the training with some of their available pre-trained weights.

If you want to reduce overfit, you may want to try more aggressive data augmentation, and other skills used in reducing overfit in training.

I don't think using higher resolution will help. The original LINEMOD dataset is not of very high resolution, and the cnn methods usually resize the image anyway.

nihar0602, 2021-04-22:
@F2Wang 

First of all thanks for this wonderful setup for creating the dataset.

Can you let me know what tool I need to use to augment the data? Any help would be great. 
Thanks!