F2Wang/ObjectDatasetTools #26: Citation

SirLamboJr, 2019-12-11:
Hi,

first of, not really an issue. I found your code really helpfull for understanding the singleshotpose and it allowed me to test their recognition on my own dataset. I want to cite this toolbox in the project I'm working on and wanted to ask if you had any preferences on that.

F2Wang, 2019-12-11:
Hi,

I am glad to hear that you found the repository useful. I have never cited a GitHub repository not related to a research work before, maybe you can try something like this: 

Fan Wang, ObjectDatasetTools, (2019), GitHub repository, https://github.com/F2Wang/ObjectDatasetTools

Thanks,
Fan

SirLamboJr, 2019-12-16:
Sorry for the wait,

but that works for me! Thanks for the reply!

closed 2019-12-16 by SirLamboJr