F2Wang/ObjectDatasetTools #27: long RECORD_LENGTH generate invalid .ply file

MrLuer, 2019-12-18:
I want to make a big dataset, so I set the RECORD_LENGTH a big value(300), the step 4 generate a invalid ply file like picture below. may be  some parameter need to be adjust?

F2Wang, 2020-01-04:
Hi, sorry about that, but I think a big RECORD_LENGTH will generally cause a bad final alignment (not to mention an extremely long processing time), as my code performs pose graph optimization only with the neighboring frames (no global loop closure), so errors will accumulate as the record length gets longer.

MrLuer, 2020-01-08:
thank you for replay，
My purpose is for the robot to grab the target, so the data set must cover as many shooting angles as possible, which determines that there must be a large RECORD_LENGTH . So is there any other way to achieve my purpose？

closed 2020-01-08 by MrLuer