F2Wang/ObjectDatasetTools #29: Some doubts about the code from create_label_files.py

zhangxiaodi, 2019-12-24:
[image]

I have some questions about "Tform = mesh.apply_obb()". What does it 'Tform' mean in this step. And why the final variable 'points_original' should convert by trimesh.transformations.transform_points?

Maybe my problem should be solved by query AABB and OBB means. But when I google its means still don't know what is means.

F2Wang, 2020-01-04:
Tform is the homogeneous transformation matrix that transforms the mesh in the initial frame so that its Orientated Bounding Box(OBB) is axis aligned and OBB center is at the origin.

This step is to make sure the extreme points are defined as the 8 corners of the OBB of the mesh.

closed 2020-03-04 by F2Wang

guohaot, 2024-05-30:
> Tform is the homogeneous transformation matrix that transforms the mesh in the initial frame so that its Orientated Bounding Box(OBB) is axis aligned and OBB center is at the origin.
> 
> This step is to make sure the extreme points are defined as the 8 corners of the OBB of the mesh.

I looked at the code for apply_obb, and I understand it as follows: apply_obb establishes a new oriented bounding box (obb) coordinate system, and rearranges the point cloud within the .ply file according to this obb coordinate system. Additionally, apply_obb returns the pose transformation matrix of the obb coordinate system relative to the camera coordinate system (because initially, the coordinate system of the .ply file is based on the camera coordinate system). This indirectly provides the initial pose of the object relative to the camera in the first frame.