F2Wang/ObjectDatasetTools #30: How to compute matrix transformation from two frames of object ?

IaroslavS, 2020-01-03:
How to compute matrix transformation (4x4) from two frames of object as an annotation .npy ?

F2Wang, 2020-01-04:
The relative transforms of all frames in regards to the first frame is saved as transforms.npy

IaroslavS, 2020-01-04:
> The relative transforms of all frames in regards to the first frame is saved as transforms.npy

Script "compute_gt_poses.py" computes transforms from clouds of points, right ?
I want to make pose annotations from keypoints for PVNet. This PVNet needs dataset formed like this:
├── /path/to/dataset
│   ├── model.ply
│   ├── camera.txt
│   ├── rgb/
│   │   ├── 0.jpg
│   │   ├── ...
│   │   ├── 1234.jpg
│   │   ├── ...
│   ├── mask/
│   │   ├── 0.png
│   │   ├── ...
│   │   ├── 1234.png
│   │   ├── ...
│   ├── pose/
│   │   ├── 0.npy
│   │   ├── ...
│   │   ├── 1234.npy
│   │   ├── ...
│   │   └──

Each .npy file contains transoframtion matrix 3x4 (not 4x4). What it can be ?
For example file "pose0.npy" for category "cat" looks like this:
array([[ 0.0950661 ,  0.983309  , -0.155129  ,  0.0717302 ],
       [ 0.741596  , -0.173913  , -0.647911  , -0.14907229],
       [-0.664076  , -0.0534489 , -0.745752  ,  1.0606388 ]],
      dtype=float32)

F2Wang, 2020-01-05:
Hi, I haven't tried PVnet but it seems like they removed the last [0,0,0,1] column from the homogeneous transformation representation. Instead of saving the transform as it is, saves it as transform[:3]

ishangupta3, 2020-02-18:
Were you able to use this repository to create a custom dataset for PVnet successfully ?

IaroslavS, 2020-02-18:
> Were you able to use this repository to create a custom dataset for PVnet successfully ?

No, I didn't try. I realized that PVNet uses only one class in training and stopped working with PVNet.

Hudson1996, 2021-05-26:
Is the content of the first three lines of ‘transforms.npy’ arranged like [R|T] ？

QRayMoor, 2022-03-31:
>   Do you know that now? Is the same thing?

QRayMoor, 2022-03-31:
> Is the content of the first three lines of ‘transforms.npy’ arranged like [R|T] ？

@Hudson1996 Do you know that now? Is the same thing? Or how to fix it? Please help

bckaiwang, 2022-10-05:
@IaroslavS  Could you tell me how did you get the camera.txt / diameter.txt / diameter.txt files?

IaroslavS, 2022-10-05:
> @IaroslavS Could you tell me how did you get the camera.txt / diameter.txt / diameter.txt files?

Hi, I've given up with this repo long ago, it was during one competition.