F2Wang/ObjectDatasetTools #32: Cannot run create_label_files.py

TommasoBendinelli, 2020-02-04:
At line 98 of create_label_files I am getting the following error:

Traceback (most recent call last):
  File "create_label_files.py", line 98, in <module>
    Tform = mesh.apply_obb()
  File "/usr/local/lib/python3.6/dist-packages/trimesh/base.py", line 2066, in apply_obb
    matrix = self.bounding_box_oriented.primitive.transform
  File "/usr/local/lib/python3.6/dist-packages/trimesh/caching.py", line 99, in get_cached
    value = function(*args, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/trimesh/parent.py", line 124, in bounding_box_oriented
    to_origin, extents = bounds.oriented_bounds(self)
  File "/usr/local/lib/python3.6/dist-packages/trimesh/bounds.py", line 132, in oriented_bounds
    if hasattr(obj, 'convex_hull'):
  File "/usr/local/lib/python3.6/dist-packages/trimesh/caching.py", line 99, in get_cached
    value = function(*args, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/trimesh/base.py", line 1995, in convex_hull
    hull = convex.convex_hull(self)
  File "/usr/local/lib/python3.6/dist-packages/trimesh/convex.py", line 53, in convex_hull
    qhull_options=qhull_options)
  File "qhull.pyx", line 2428, in scipy.spatial.qhull.ConvexHull.__init__
  File "qhull.pyx", line 357, in scipy.spatial.qhull._Qhull.__init__
scipy.spatial.qhull.QhullError: QH7023 qhull option warning: unknown 'Q' qhull option 'Qn', skip to next space
QH6035 qhull option error: see previous warnings, use 'Qw' to override: 'qhull i QJn Qt Pp QbB' (last offset 9)

While executing:  | qhull i QJn Qt Pp QbB
Options selected for Qhull 2019.1.r 2019/06/21:
  run-id 428483682  incidence  Qtriangulate  Pprecision-ignore
  QbBound-unit-box 0.5  _maxoutside  0

TommasoBendinelli, 2020-02-04:
resolve by downgrading to scipy 1.3.3, but now I am getting the following error: 
test3 is assigned class label 0.
 57%|██████████████████████████████████████████████████████████████████▉                                                   | 669/1179 [00:36<00:23, 21.41it/s]Traceback (most recent call last):
  File "create_label_files.py", line 156, in <module>
    cnt = max(contours, key=cv2.contourArea)
ValueError: max() arg is an empty sequence

TommasoBendinelli, 2020-02-05:
I bypass the error by adding the following:
try:
                cnt = max(contours, key=cv2.contourArea)
            except:
                pass 
I suspect that the error is because in some frames there is not object.

F2Wang, 2020-02-17:
Hi, thank you for the comments. Ideally you should keep the object within frame at all time, otherwise it may cause unexpected errors.

closed 2020-02-17 by F2Wang