F2Wang/ObjectDatasetTools #35: Question about step 5

JacksonnnTan, 2020-02-23:
1. The registered pointcloud (manually) mentioned in step 5 is it the same thing generated by the register_scene.py ? (But it seems that register_scene.py already will generate the mesh )

2. If the answer is yes in previous question, then can we just use the result from register_segmented.py without the need to do step 5 ?

3. If we really need to use meshlab, is it ok to just use the same setting as what you did in your youtube video ?

Thank you

F2Wang, 2020-02-24:
Hi,

Yes step 5 manually process the pointcloud generated by register_scene.py in meshLab, you cannot skip this step, unfortunately.

The video is a rough guideline and it may or may not work for your case. If you cannot get a good surface reconstruction following the steps, you should check out other meshlab tutorials.

Thanks,
Fan

JacksonnnTan, 2020-02-24:
Thank you, but if we use register_segmented.py instead of egister_scene.py, can we skip step 5 ?

F2Wang, 2020-02-24:
You will need to manually process it even with register_segmented, registered_segmented generate registered points, in step 5 you convert the points to a triangular mesh.

JacksonnnTan, 2020-02-24:
Ok got you thanks

closed 2020-03-04 by F2Wang