F2Wang/ObjectDatasetTools #38: Trimesh Version, AttributeError: 'PointCloud' object has no attribute 'visual'

ishangupta3, 2020-03-02:
I believe I am not using the same version of Trimesh as the authors, as I am using their sample provided dataset of a timer. Would this issue be because of the current version of Trimesh I do have, if so which would be the correct one to install?

(boom) bash-3.2$ python create_label_files.py LINEMOD/timer
timer is assigned class label 0.
Traceback (most recent call last):
  File "create_label_files.py", line 100, in <module>
    mesh.export(file_obj = folder + folder[8:-1] +".ply")
  File "/Users/ishangupta/.pyenv/versions/boom/lib/python3.6/site-packages/trimesh/points.py", line 620, in export
    **kwargs)
  File "/Users/ishangupta/.pyenv/versions/boom/lib/python3.6/site-packages/trimesh/exchange/export.py", line 68, in export_mesh
    export = _mesh_exporters[file_type](mesh, **kwargs)
  File "/Users/ishangupta/.pyenv/versions/boom/lib/python3.6/site-packages/trimesh/exchange/ply.py", line 150, in export_ply
    if mesh.visual.kind == 'vertex' and encoding != 'ascii':
AttributeError: 'PointCloud' object has no attribute 'visual'

F2Wang, 2020-03-03:
Hi, I don't believe this is a trimesh version issue, did you skip step 5 where you convert the registered PointCloud into a triangular mesh?

ishangupta3, 2020-03-04:
@F2Wang You were right, thanks.

closed 2020-03-04 by ishangupta3