F2Wang/ObjectDatasetTools #39: Mask vs RGB image

ishangupta3, 2020-03-04:
Below is a generated mask with the respective RGB image, the mask does not seem to be a pixel perfect of the jpeg. Is there something I am doing wrong while recording with the Real Sense camera, should I be holding the camera in a certain way or moving it in a specific way?

[image]
[image]

F2Wang, 2020-03-04:
It looks like the generated mesh is very inaccurate. This could either come from misalignment during the registration process or a failed meshing process from the registered pointcloud.

Did you go through step 5 manually, or did you use register_segmented to generate the mesh automatically? Try running register_scene to see if the registered pointcloud looks correct aligned first.

It should be noted that the depth of the item you are trying to reconstruct cannot be captured with Realsense cameras very well, as structured light based depth camera cannot see most shades of black, as well as transparent and reflective objects.

You are always welcomed to send me your dataset to faninedinburgh@gmail.com so I have a better idea of what went wrong.

Thanks.

ishangupta3, 2020-03-04:
Thank you for the detailed response. I will try with a different object.

closed 2020-03-04 by ishangupta3