F2Wang/ObjectDatasetTools #40: Confirmation of Extraction of R,t Matrix

ishangupta3, 2020-03-04:
pose = np.load(folder)
        print(pose.shape)  # returns a 4 x 4 matrix

        pose = pose[:3]  # this drops the  [0,0,0,1] column

        R = pose.T[:3]
        t = pose.T[3]

I just want to confirm, if the above would be the correct way to extract the R,t matrx from each of the numpy transform folders. I am basing it off this transform matrix found in match_ransac:

 transform = [[R[0][0],R[0][1],R[0][2],t[0]],
                     [R[1][0],R[1][1],R[1][2],t[1]],
                     [R[2][0],R[2][1],R[2][2],t[2]],
                     [0,0,0,1]]

F2Wang, 2020-03-05:
Hi, the translation looks correct but you transposed the rotation matrix
for R, do pose[:3, :3] instead.

Thanks

ishangupta3, 2020-03-05:
Got it, thanks @F2Wang.

closed 2020-03-05 by ishangupta3