F2Wang/ObjectDatasetTools #42: Changing of RGBD data

ishangupta3, 2020-03-22:
I would like to change the code where I can keep the real sense camera stationary and take a photo where each photo consists of a different pose of the object.

Would the rest of the steps still work with the above changes?

F2Wang, 2020-03-22:
It will not work unfortunately , the pose of the marker and the object need to fixed while filming.

andgitchang, 2020-10-28:
Hi @F2Wang  
Following up this issue, if I would like to customize human 3d pose dataset using this repo, the pose of human must also be fixed throughout filming? If so, do you have any insightful suggestion how can I generalize singleshotpose/ObjectDatasetTools to human 3d _static_ pose estimation? Where the _static_ pose e.g. standing, holding things in hands. Look forward to your reply and thanks in advance.