F2Wang/ObjectDatasetTools #43: whereregister_scene.py

MengHao666, 2020-04-07:


F2Wang, 2020-04-07:
It's in the camera's coordinate frame where the camera is at its initial pose when starts filming.

If you know the transform from the camera's coordinate frame at its initial position to the robot base frame, you can transform the generated pointcloud to robot coordinate to act on,

MengHao666, 2020-04-07:
Thanks for your fast reply.
Other 2 question:
  1.  I found it it take a lot of time to run "python compute_gt_poses.py LINEMOD/sugar" 
 step, so to achieve real-time ,how should i accelerate the processes:  "Obtain frame transforms" and "Register all frames and create a mesh for the registered scene."? For me ,I just want to get the object's pointclouds which maybe not too precise but can show the overall rough distribution.Are there any parameters i can change to achieve a "downsampling"?

2. The code now is useful for single object scene ,but how should I extend it to multiple objects scene?Is it difficult to achieve?

Hopefully u could give me some suggestions.
Thank u again!

MengHao666, 2020-04-07:
Thanks for your fast reply.
Other 2 question:

I found it it take a lot of time to run "python compute_gt_poses.py LINEMOD/sugar"
step, so to achieve real-time ,how should i accelerate the processes: "Obtain frame transforms" and "Register all frames and create a mesh for the registered scene."? For me ,I just want to get the object's pointclouds which maybe not too precise but can show the overall rough distribution.Are there any parameters i can change to achieve a "downsampling"?

The code now is useful for single object scene ,but how should I extend it to multiple objects scene?Is it difficult to achieve?

Hopefully u could give me some suggestions.
Thank u again!

F2Wang, 2020-04-10:
Hi sorry for the delay.

For the first question, yes this code is not for realtime usage. To speed up, you can modify parameters in registration parameters under config (e.g. increase VOXEL_SIZE for faster ICP, increase LABEL_INTERVAL and RECONSTRUCTION_INTERVAL so that you don't compute transform for every single frame, while RECONSTRUCTION_INTERVAL should be multiples of LABEL_INTERVAL)n, etc

You can also use other SLAM projects with GPU acceleration, like KinectFusion, dynamicFusion. Also if you mount the camera on the robot and scan the scene with the robot arm movement, you should know the relative transforms from your robot configuration.

For the second object, if your intention is to get a single pointcloud with all objects in, then you don't need to change anything about the code. If you intend to get multiple pointclouds belonging to every single object in the scene, you need to add a segmentation step, if items are placed in isolation, PCL's region growing and clustering should work quite well.

zhangxiaodi, 2020-04-26:
Maybe you should know the hand-eye-calibrate process. And it can transform the camera coordinate to the robot arm.@MengHao666

MengHao666, 2020-04-26:
Sorry for late response.
Very grateful for suggestions from  @F2Wang and @zhangxiaodi.
I would try the methods u suggested, and give the feedback later.
Thank u again!

closed 2020-04-26 by MengHao666