F2Wang/ObjectDatasetTools #44: about register_scene.py

shanniruo, 2020-04-13:
when i run register_scene.py ,The reconstruction result is relatively bad,I think it's the reason for the video，but i don't get a good result after several attempts , so about make a viedo,Do you have any guidance？or another reason?

F2Wang, 2020-04-13:
Hi, do you have pictures of the bad reconstruction? It's hard to tell what the problem is without looking at one.

Also, you are welcomed to send me the sequence to faninedinburgh@gmail.com so that I can take a look.

Thanks

shanniruo, 2020-04-14:
@F2Wang I've already sent it

F2Wang, 2020-04-16:
Hi,

Sorry it took a while, I think I finally found what the problem is.
You have 2 aruco markers with id=3 and 2 aruco markers with id=5  displayed, this will result in false alignment with cues from aruco marker. Would you make sure that you have none-duplicate markers?
Also, it appears that you are using the realsense D series camera, the depth quality is not as good as the SR300 series, so it's quite unlikely that you will have a model with perfect quality.

Let me know if the problem persists.

shanniruo, 2020-04-16:
I've solved it .thank you

closed 2020-04-19 by F2Wang