F2Wang/ObjectDatasetTools #51: RunTimeError in Step6

xiaoru222, 2020-06-12:
@F2Wang   
 when I try "python create_label_files.py all", I got an error below

timer is assigned class label 0.
  0%|                                                                                                  | 0/55 [00:00<?, ?it/s] 
Traceback (most recent call last):
  File "create_label_files.py", line 128, in <module>
    sample_points = mesh_copy.sample(10000)
AttributeError: 'PointCloud' object has no attribute 'sample'

Hope someone gives some hints

nihar0602, 2021-04-12:
@xiaoru222 were you able to solve this?

cgm6666, 2021-04-14:
hi, did you solve the problem

nihar0602, 2021-04-22:
@chengm0-0, 
HI Cheng, I was able to solve this issue.

hyoon970, 2021-04-26:
@nihar0602 
Hello. I got the same problem. How could you solve it?

qingweihk, 2021-05-31:
> @chengm0-0,
> HI Cheng, I was able to solve this issue.

HI Nihar,I got the same question.Could you please help me ?

hyoon970, 2021-05-31:
> > @nihar0602
> > Hello. I got the same problem. How could you solve it?
> 
> Hi hyoon970,I got the same question, have you solved the problem?

Hello, I don't remember it exactly, but I solved it.

After ply processing with Meshlab, may be you saved the ply file with another name, not 'registeredScene.ply'.

But create_label_files.py read 'registeredScene.ply' file. 

So you should change the original ply file's name into another one, like 'registeredScene_orogonal.ply' and rename the processed ply file as 'registeredScene.ply'.

If you can't solve your problem with solution above, reply to me anytime.

Thanks and regards,

H.Yoon

qingweihk, 2021-05-31:
> > > @nihar0602
> > > Hello. I got the same problem. How could you solve it?
> > 
> > 
> > Hi hyoon970,I got the same question, have you solved the problem?
> 
> @qingweihk
> 
> Hello, I don't remember it exactly, but I solved it.
> 
> After ply processing with Meshlab, may be you saved the ply file with another name, not 'registeredScene.ply'.
> 
> But create_label_files.py read 'registeredScene.ply' file.
> 
> So you should change the original ply file's name into another one, like 'registeredScene_original.ply' and rename the processed ply file as 'registeredScene.ply'.
> 
> If you can't solve your problem with solution above, reply to me anytime.
> 
> Thanks and regards,
> 
> H.Yoon

Thanks very much.

I wonder after Meshlab process:the  'registeredScene.ply' should be  'registeredScene.obj? Cause The end product is a triangular mesh instead of the registered pointcloud( the repo's instruction at  part 5)

hyoon970, 2021-05-31:
> Thanks very much.
>
> I wonder after Meshlab process:the 'registeredScene.ply' should be 'registeredScene.obj? Cause The end product is a triangular mesh instead of the registered pointcloud( the [...]

No, I used ply file and succeed. I don't know much about Meshlab, so I just followed the Youtube video the author uploaded.

qingweihk, 2021-05-31:
> No, I used ply file and succeed. I don't know much about Meshlab, so I just followed the Youtube video the author uploaded.

Thanks very much!

lzhhh6161, 2021-06-09:
> Thanks very much!

hi,I got the same question, have you solved the problem?

hyoon970, 2021-06-11:
@lzhhh6161 
Solutions above do not work?

forbeauty, 2022-08-31:
replace the following sentence:
- sample_points = mesh_copy.sample(10000)
into:
            dellist = [j for j in range(0, len(mesh_copy.vertices))]
            dellist = random.sample(dellist, len(mesh_copy.vertices) - 10000)
            sample_points = np.delete(mesh_copy.vertices, dellist, axis=0)

jefferson-norberto, 2022-09-09:
Thanks. It's working here.