F2Wang/ObjectDatasetTools #53: Issue with step 6

KatharinaSchmidt, 2020-07-04:
Hello, 

i tried to run step 6 with `python create_label_files.py all` and generate labels for the example class "timer". 
But I get the following error message: 

timer is assigned class label 0.
  0%|                                                                                          | 0/600 [00:00<?, ?it/s]
Traceback (most recent call last):
  File "create_label_files.py", line 154, in <module>
    _, contours, _ = cv2.findContours(thresh.copy(), cv2.RETR_EXTERNAL,
ValueError: not enough values to unpack (expected 3, got 2)

And my triangular mesh (generated with `register_segmented.py`) semms to look a little bit weird: 
[image]

Even if I process the registered pointcloud manually (as described in step 5) I get that error message above. 

Can you please give me an advice how to solve this? 
Thanks

naaths, 2020-07-15:
delete first
_,

Sciroccogti, 2021-03-05:
I got exactly the same mesh XD