F2Wang/ObjectDatasetTools #54: Issue with step 2: compute_gt_poses.py is getting killed after certain percentage.

redwanulsourav, 2020-07-19:
I am a beginner in preparing 3d datasets. But while performing step 2, the process is getting killed at certain percentage with only the message: killed.

Also I wanted to know if there is any certain order of placing the markers? I placed the markers around the object in a rectangular fashion with no order maintained.

Sciroccogti, 2021-03-05:
There seems to be no order of these markers.

GaoYifan0523, 2021-06-30:
the process is getting killed at certain percentage with only the message: killed.  I have the same problem. Have you solved your problem？

redwanulsourav, 2021-07-12:
> the process is getting killed at certain percentage with only the message: killed. I have the same problem. Have you solved your problem？

No I could not solve it. I left it cause I could not solve the issue.

MaxShi007, 2021-12-09:
Because the out of mem. I tried compute_gt_poses.py on my virtual machine(I assigned 4G ram), its killed after 5%. then I tried it in my server, its done(32g VIRT,8.8g RES)

delonixsen, 2022-03-21:
increase swap space, it helps me 

https://askubuntu.com/questions/178712/how-to-increase-swap-space/178726#178726