F2Wang/ObjectDatasetTools #59: Adding my own model

luigifaticoso, 2021-02-24:
Hello, 
Thank you for your work.
I have a question, is it possible to add my own object to the scene instead of creating a mesh from the reconstructed scene? 
How can I do it? 
thank you

luigifaticoso, 2021-03-02:
I managed to do it by adding my .ply model to the scene and aligning it with the real one. By doing so I skipped the 5th step.

closed 2021-03-02 by luigifaticoso

nihar0602, 2021-04-12:
Hi @luigifaticoso,
Thanks for this query.  

Did you create your .ply model separately? 
What tools you used for that and while creating .ply, did you keep the camera stationary and the object rotating?

luigifaticoso, 2021-04-12:
Hello @nihar0602 

In my case, I already had the 3D model of the object. I sampled some points on it since I didn't have vertices on my 3D model and converted it to point cloud using cloud compare.

nihar0602, 2021-04-12:
Hi @luigifaticoso 
Thank you for your response.

bj-bai, 2022-03-30:
hi@luigifaticoso
Excuse me ，How to solve the scale problem between model and real scene?The scene is too small and the model is too large