F2Wang/ObjectDatasetTools #64: register_segmented.py result is not the same with README.md's result.

Matthew-CQ, 2021-04-06:
Hi Wang,

Thank you for your great work. I got a different segmented result after running **register_segmented.py** using the default timer dataset. Could you please help t me to solve it? Thanks.
[image]

Regards,
Quan

l584825739, 2023-08-18:
excuseme,  You run the code successfully, but when i run register_segmented.py, 
get a error: TypeError: Improper input: func input vector length N=4 must not exceed func output vector length M=0,

Have you ever had this problem?

Harrydsk, 2024-02-29:
> excuseme, You run the code successfully, but when i run register_segmented.py, get a error: TypeError: Improper input: func input vector length N=4 must not exceed func output vector length M=0,
> 
> Have you ever had this problem?

excuse me.
did u  solve this problem