F2Wang/ObjectDatasetTools #65: compute_gt_poses.py issue

hyoon970, 2021-04-15:
Hello,

I'm trying to make custom dataset.

While doing step 3, when I use compute_gt_poses.py, there are two issues.

below is my current state.
___________________________________________________________________________________________________________________

PS C:\Users\user\Desktop\ObjectDatasetTools-master> python compute_gt_poses.py all
LINEMOD\febriz\
Full registration ...
 34%|███████████████▎                             | 75/221 [18:48<14:43,  6.05s/it]C
**:\Users\user\Desktop\ObjectDatasetTools-master\registration.py:204: RuntimeWarning:
invalid value encountered in double_scalars
  RMSE = np.sum(error[np.argpartition(error, k)[:k]])/k**
 98%|███████████████████████████████████████████▏| 217/221 [25:39<00:04,  1.15s/it][
Open3D WARNING] [KDTreeFlann::SetRawData] Failed due to no data.
 98%|███████████████████████████████████████████▏| 217/221 [25:40<00:28,  7.10s/it]
Traceback (most recent call last):
  **File "compute_gt_poses.py", line 351, in <module>
    pose_graph = full_registration(path, max_correspondence_distance_coarse,
  File "compute_gt_poses.py", line 180, in full_registration
    transformation_icp, information_icp = icp(
  File "C:\Users\user\Desktop\ObjectDatasetTools-master\registration.py", line 49, i
n icp
    icp_coarse = pipelines.registration.registration_icp(source, target,
RuntimeError: [Open3D ERROR] TransformationEstimationPointToPlane and Transformation
EstimationColoredICP require pre-computed normal vectors for target PointCloud.** 
_________________________________________________________________________________________________________

I followed 'Pull request' to fix some codes.

However, there are still some problems.

Anyone who saw this issue?

Please help me.

Kinds and Regards,

H.Yoon

hyoon970, 2021-04-15:
When I use 'timer', there is no problem like above.

Therefore, I could find that the problem is from my images.

After deleting some images(75, 76), although there was still 'invalid value encountered in double_scalars' problem, I could make transforms.npy anyway.

Thanks.

closed 2021-04-15 by hyoon970

QRayMoor, 2022-03-31:
@hyoon970 Hi, do you know about transforms.npy?   I don't understand the format and meaning it represents.
transform = [[R[0][0],R[0][1],R[0][2],t[0]],       or     [[R[0][0], R[1][0], R[2][0], t[0]],
                    [R[1][0],R[1][1],R[1][2],t[1]],                 [R[0][1], R[1][1], R[2][1], t[1]],
                    [R[2][0],R[2][1],R[2][2],t[2]],                 [R[0][2], R[1][2], R[2][2], t[2]],
                    [0,         0,        0,          1]]                   [0,         0,        0,          1]]

QRayMoor, 2022-03-31:
@hyoon970 Hi, do you know about transforms.npy?   I don't understand the format and meaning it represents.
transform = [[R[0][0],R[0][1],R[0][2],t[0]],      
                    [R[1][0],R[1][1],R[1][2],t[1]],                
                    [R[2][0],R[2][1],R[2][2],t[2]],                 
                    [0,         0,        0,          1]]          
or
transform = [[R[0][0], R[1][0], R[2][0], t[0]],
                    [R[0][1], R[1][1], R[2][1], t[1]],
                    [R[0][2], R[1][2], R[2][2], t[2]],
                    [0,         0,        0,          1]]