F2Wang/ObjectDatasetTools #74: Succesful Method for DenseFusion

delonixsen, 2022-07-09:
I'm planning on making this version of ObjectDatasetTools to have an extension for DenseFusion, I'm sucessfully reassembling singleshotpose groundtruth label to DenseFusion camera pose gt.yml, Thanks for @F2Wang for making this Third Party Program, which is very helpful!

closed 2022-07-09 by delonixsen

jasmine202106, 2022-08-19:
hi,
would you please tell me how to add an extension for densefusion?
how to generate gt.yml, info.yml and models_info.yml?

thanks very much!

JayLuoShui, 2022-08-28:
> hi, would you please tell me how to add an extension for densefusion? how to generate gt.yml, info.yml and models_info.yml?
> 
> thanks very much!

Hi，I want to know if this problem is solved

zzzcd0x, 2022-08-29:
> 

Hi, 
I'm confused about how to generate gt.yml and info.yml, could you please tell me how to do that?

JayLuoShui, 2022-09-04:
> > 
> 
> Hi, I'm confused about how to generate gt.yml and info.yml, could you please tell me how to do that?

Hi，I think you can try  bop_tolkit

zzzcd0x, 2022-09-08:
> > > 
> > 
> > 
> > Hi, I'm confused about how to generate gt.yml and info.yml, could you please tell me how to do that?
> 
> Hi，I think you can try bop_tolkit
Sorry for replying so late. I have already done that with it. Thank you so much.

8umpk1n, 2022-11-10:
> > 嗨，你能告诉我如何为密集融合添加扩展名吗？如何生成 gt.yml、info.yml 和 models_info.yml？
> > 非常感谢！
> 
> 你好，我想知道这个问题是否解决了

hi,请问是[bop_toolkit]吗 这个文件该如何使用呢，谢谢