F2Wang/ObjectDatasetTools #78: NameError: name 'registration' is not defined

sowmyakavali, 2022-11-08:
Hi @paperstiger @F2Wang !
    Please look into this issue,
```
           ```
Traceback (most recent call last):
  File "compute_gt_poses.py", line 351, in <module>
    pose_graph = full_registration(path, max_correspondence_distance_coarse,
  File "compute_gt_poses.py", line 152, in full_registration
    pose_graph = registration.PoseGraph()
NameError: name 'registration' is not defined
```
```
That registration module itself not imported and even don't know that where it is defined.
can you please check it ?

sowmyakavali, 2022-11-08:
> Hi @paperstiger @F2Wang ! Please look into this issue,
>
> ```
>            ```
> Traceback (most recent call last):
>   File "compute_gt_poses.py", line 351, in <module>
>     pose_graph = [...]

 I can able to solve this by the following way
first do import open3d in this way
   `from open3d import pipelines`
  then replace 
             `registration.get_information_matrix_from_point_clouds()`
   with
          `pipelines.registration.get_information_matrix_from_point_clouds()`

tiny-ych-wolf, 2023-05-04:
You can  simply solve this problem by doing those steps: uninstall opencv-python 4.7 and install opencv 4.6.0.66

AqibFarooq, 2023-05-11:
> You can simply solve this problem by doing those steps: uninstall opencv-python 4.7 and install opencv 4.6.0.66
this didnt solve the issue

wangzihanggg, 2024-01-07:
> I can able to solve this by the following way first do import open3d in this way `from open3d import pipelines` then replace `registration.get_information_matrix_from_point_clouds()` with [...]

Thank u very much! Tha saved me.